andyboeh/esphome-dmx512 #1: Validate and compile fails

jsimonetti, 2021-12-02:
Hello.
First of all, thank you for making this custom component. It is exactly what I needed.

I am having a problem getting this to work using this yaml:
```
packages:
  wifi: !include common/wifi.yaml
  device_base: !include common/device_base.yaml
  
external_components:
  - source: github://andyboeh/esphome-dmx512

esphome:
  platform: ESP32
  board: nodemcu-32s
        
uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: 5
  stop_bits: 2
  
dmx512:
  id: dmx_0
  uart_id: uart_bus
  enable_pin: 33
  tx_pin: 5
  uart_num: 1

output:
  - platform: dmx512
    channel: 1
    universe: dmx_0
    id: dmx_0_ch1
  - platform: dmx512
    channel: 2
    universe: dmx_0
    id: dmx_0_ch2
  - platform: dmx512
    channel: 3
    universe: dmx_0
    id: dmx_0_ch3
  - platform: dmx512
    channel: 4
    universe: dmx_0
    id: dmx_0_ch4
  - platform: dmx512
    channel: 5
    universe: dmx_0
    id: dmx_0_ch5
  - platform: dmx512
    channel: 6
    universe: dmx_0
    id: dmx_0_ch6
  - platform: dmx512
    channel: 7
    universe: dmx_0
    id: dmx_0_ch7
  - platform: dmx512
    channel: 8
    universe: dmx_0
    id: dmx_0_ch8
  - platform: dmx512
    channel: 9
    universe: dmx_0
    id: dmx_0_ch9
  - platform: dmx512
    channel: 10
    universe: dmx_0
    id: dmx_0_ch10
  - platform: dmx512
    channel: 11
    universe: dmx_0
    id: dmx_0_ch11
  - platform: dmx512
    channel: 12
    universe: dmx_0
    id: dmx_0_ch12

light:
  - platform: rgb
    name: "RGB1"
    red: dmx_0_ch1
    green: dmx_0_ch2
    blue: dmx_0_ch3
  - platform: rgb
    name: "RGB2"
    red: dmx_0_ch4
    green: dmx_0_ch5
    blue: dmx_0_ch6
  - platform: rgb
    name: "RGB3"
    red: dmx_0_ch4
    green: dmx_0_ch5
    blue: dmx_0_ch6
  - platform: rgb
    name: "RGB4"
    red: dmx_0_ch4
    green: dmx_0_ch5
    blue: dmx_0_ch6
```

When I try to validate the config I get the error:
```
  File "/data/persist/esphome/.esphome/external_components/9d3a21bd/components/dmx512/__init__.py", line 20, in _declare_type
    if CORE.using_arduino:
AttributeError: 'EsphomeCore' object has no attribute 'using_arduino'
```
I am unsure what this error means. Could you assist with this?

If I comment out the `if CORE.using_arduino` statement, the validation works, however compiling fails with this error:
```
INFO Reading configuration /data/persist/esphome/woonkamer_rgbleds.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight saving time from 27 March 02:00:00 to 30 October 03:00:00
INFO Generating C++ source...
/nix/store/k44dvza23wpjbry43qd985r07li26lpa-esphome-1.18.0/lib/python3.8/site-packages/esphome/core/__init__.py:678: RuntimeWarning: coroutine 'to_code' was never awaited
  next(task)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
ERROR Circular dependency detected!
```

I hope you have any idea where I should start troubleshooting this.

andyboeh, 2021-12-02:
Your ESPHome version is too old. Support for different cores was added only recently. I'm running 2021.11.3.

jsimonetti, 2021-12-02:
Thank you! Updating to 2021.11.4 fixed the compilation.

closed 2021-12-02 by jsimonetti