andyboeh/esphome-dmx512 #2: Allow GPIO-labeled pins in config

nagyrobi, 2022-01-15:
Config:

```
uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: GPIO5
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  enable_pin: GPIO33
  tx_pin: GPIO5
  uart_num: 1

output:
  - platform: dmx512
    channel: 1
    universe: dmx
    id: dmx_1

light:
  - platform: monochromatic
    name: "Test Lights"
    output: dmx_1
    id: light_test
    default_transition_length: 2s
```

Error at build:
```
INFO Reading configuration /config/esphome/esp32-test-1-dmx.yaml...
INFO Cloning https://github.com/andyboeh/esphome-dmx512.git@None
Failed config

dmx512: [source /config/esphome/esp32-test-1-dmx.yaml:79]
  - [source /config/esphome/esp32-test-1-dmx.yaml:80]
    id: dmx
    uart_id: uart_bus
    enable_pin: GPIO33
    
    Expected integer, but cannot parse gpio5 as an integer.
    tx_pin: GPIO5 [source /config/esphome/esp32-test-1-dmx.yaml:83]
    uart_num: 1
```

To avoid confusion, [ESPHome added](https://esphome.io/guides/configuration-types.html) `GPIO` string to the pins to differentiate pin numbering of the MCU from pin numbering of various boards.
Please allow us in this component to use the same pin addressing scheme.

andyboeh, 2022-01-15:
Please send a PR.

nagyrobi, 2022-01-15:
I see.

commit 7d50019e (2022-01-16): Fix #2 and add more documentation
closed 2022-01-16 by andyboeh