andyboeh/esphome-dmx512 #3: How to set DMX fixture address?

nagyrobi, 2022-01-15:
Thank you very much for this component.
It works great with [ADJ VBar Pak](https://d295jznhem2tn9.cloudfront.net/ItemRelatedFiles/8659/vbar_pak.pdf) fixtures using the config below. It is possible to control effects too!

The only issue is that there's no way to specify the DMX address of the light, as we have 4 or them, daisy chained with standard DMX cable. Please help me how to implement addressing to set up the lights.

```
uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: 5
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  enable_pin: 33
  tx_pin: 5
  uart_num: 1
  force_full_frames: true

output:
- platform: dmx512
  channel: 1
  universe: dmx
  id: vbar_red
- platform: dmx512
  channel: 2
  universe: dmx
  id: vbar_green
- platform: dmx512
  channel: 3
  universe: dmx
  id: vbar_blue
- platform: dmx512
  channel: 4
  universe: dmx
  id: vbar_amber
- platform: dmx512
  channel: 5
  universe: dmx
  id: vbar_master
- platform: dmx512
  channel: 6
  universe: dmx
  id: vbar_effect_level
- platform: dmx512
  channel: 7
  universe: dmx
  id: vbar_effect_select
- platform: dmx512
  channel: 8
  universe: dmx
  id: vbar_effect_mode

number:
- platform: template
  name: "VBar Pak Effect Level"
  icon: mdi:gauge-low
  id: vbar_effect_level_num
  optimistic: true
  min_value: 0
  max_value: 255
  step: 1
#  mode: box
  restore_value: true
  set_action:
    - lambda: |-
        id(vbar_effect_level).set_level(x / 255);

- platform: template
  name: "VBar Pak Effect Mode"
  icon: mdi:gauge
  id: vbar_effect_mode_num
  optimistic: true
  min_value: 0
  max_value: 255
  step: 1
  mode: box
  restore_value: true
  set_action:
    - lambda: |-
        id(vbar_effect_mode).set_level(x / 255);

select:
- platform: template
  name: VBar Pak Effect Mode
  optimistic: true
  restore_value: true
  icon: "mdi:led-strip-variant"
  id: vbar_effect_select_dropdown
  options:
    - "None"
    - "Color Macro"
    - "Color Change"
    - "Color Fade"
    - "Sound Active"
    - "Strobing"
  set_action:
    - lambda: |-
          if (0 == x.compare(std::string{"None"})) {
            id(vbar_effect_select).set_level(0);
            auto call_level = id(vbar_effect_level_num).make_call();
            auto call_mode = id(vbar_effect_mode_num).make_call();
            call_level.set_value(0);
            call_mode.set_value(0);
            call_level.perform();
            call_mode.perform();
          }
          if (0 == x.compare(std::string{"Strobing"})) {
            id(vbar_effect_select).set_level(0.04);
          }
          if (0 == x.compare(std::string{"Color Macro"})) {
            id(vbar_effect_select).set_level(0.3);
          }
          if (0 == x.compare(std::string{"Color Change"})) {
            id(vbar_effect_select).set_level(0.5);
          }
          if (0 == x.compare(std::string{"Color Fade"})) {
            id(vbar_effect_select).set_level(0.66);
          }
          if (0 == x.compare(std::string{"Sound Active"})) {
            id(vbar_effect_select).set_level(0.9);
            auto call = id(vbar_effect_level_num).make_call();
            call.set_value(200);
            call.perform();
          }

light:
- platform: monochromatic
  name: "VBar Pak Dimmer"
  output: vbar_master
  icon: mdi:white-balance-sunny
  default_transition_length: 2s
  restore_mode: RESTORE_DEFAULT_ON

- platform: rgbw
  name: "VBar Pak"
  red: vbar_red
  green: vbar_green
  blue: vbar_blue
  white: vbar_amber
  icon: mdi:spotlight
  default_transition_length: 0s
  color_interlock: false
  restore_mode: RESTORE_DEFAULT_ON
```

andyboeh, 2022-01-16:
I don't quite get your request. Every output has a channel and the channel is the DMX address!? In order to support multiple devices, each device must have its own unique starting address - this is even explained in your linked PDF in on pages 7/8 (DMX Mode).

nagyrobi, 2022-01-16:
Yes! Each device has a settable address (between 1 and 512). This can be set from the buttons on the light as written on page 7/8. 

For each addressed light, we have channels from 1 to 8 - these work as outputs as configured above.

This component only works when light is set to address 1 (`d.001` on the LED display). Next light is set to address 2 (`d.002`), but channel numbers are the same.

Or is there some offset number I could manually add to the channel number to reach the next addressed light?

nagyrobi, 2022-01-16:
Like this:
[image]

nagyrobi, 2022-01-16:
As a workaround, it's possible to configure the raw channel numbers by calculating:
channel (in esphome component) = address + channel (on the fixture).

andyboeh, 2022-01-16:
> As a workaround, it's possible to configure the raw channel numbers by calculating:
> channel (in esphome component) = address + channel (on the fixture).

This is exactly how it's done - that's not a workaround, but the solution :)

nagyrobi, 2022-01-16:
From the user's perspective it's a workaround to configure it like this...

Would you accept a PR to the readme to clarify this stuff as I see others run into the same...?

andyboeh, 2022-01-16:
Yes, of course.

closed 2022-01-16 by nagyrobi