andyboeh/esphome-dmx512 #4: Compile fails for ESP8266 version

nagyrobi, 2022-01-16:
Using `source: github://andyboeh/esphome-dmx512@esp8266_support` and ESPHome 2021.12.3 or 2022.1.0b2 with Home Assistant Add-On, build fails with the following error:

```
Compiling /data/esp8266-test-1-dmx/.pioenvs/esp8266-test-1-dmx/lib975/ESPAsyncTCP-esphome/AsyncPrinter.cpp.o
Compiling /data/esp8266-test-1-dmx/.pioenvs/esp8266-test-1-dmx/lib975/ESPAsyncTCP-esphome/ESPAsyncTCP.cpp.o
src/main.cpp:31:1: error: 'DMX512ESP8266' in namespace 'esphome::dmx512' does not name a type
 dmx512::DMX512ESP8266 *dmx;
 ^
src/main.cpp: In function 'void setup()':
src/main.cpp:337:3: error: 'dmx' was not declared in this scope
   dmx = new dmx512::DMX512ESP8266();
   ^
src/main.cpp:337:13: error: expected type-specifier
   dmx = new dmx512::DMX512ESP8266();
             ^
src/main.cpp:337:13: error: expected ';'
*** [/data/esp8266-test-1-dmx/.pioenvs/esp8266-test-1-dmx/src/main.cpp.o] Error 1
========================= [FAILED] Took 66.97 seconds =========================
```

ESP32 is OK.

andyboeh, 2022-01-16:
Forgot to commit some files, should be fixed now.

nagyrobi, 2022-01-16:
Tried with GPIO pins 1 or 5 on a Wemos D1 Mini and it doesn't work. I know ESP8266 has only one UART which may cause conflicts, do you have a recommendation on which GPIO to use?

nagyrobi, 2022-01-16:
Works with GPIO 2 (pin D4 on Wemos D1 Mini). Thank you!

andyboeh, 2022-01-16:
1. Please do not post new issues to other, already closed issues.
2. Exactly, it can only work with GPIO2 as this is the only hardware UART pin the ESP8266 has.

So basically, you are confirming that the implementation works on a ESP8266?

closed 2022-01-16 by andyboeh

nagyrobi, 2022-01-16:
> So basically, you are confirming that the implementation works on a ESP8266?

Yes, it works, 4 fixtures in daisy chain!

andyboeh, 2022-01-16:
Perfect, thanks!

commit 3b82f4d7 (2022-01-16): Fix #4