andyboeh/esphome-dmx512 #11: Resets

itsieber, 2022-01-30:
Salut zusammen

Sobald ich ich mehr als 35 RGBW Lichter mit je 4 Kanälen erstelle, läuft es nicht mehr oder sehr instabiel.

Irgend wer eine Idee?

andyboeh, 2022-01-30:
Please ask in English.

itsieber, 2022-01-31:
Sorry, my mistake

When i have many rgbw then the dmxcontrollort on esp32 it will go to reboots on nearly the end of the init.
After remove 5 Actors then it's run but unstable.
After remove again a few actors its run stable.

About the log its a loop-detection.

The RS485 adapter is this on the readme. With my own old esp programm without homeassist it was run without problems.

uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: 18
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: 18
  uart_num: 1

output:
  - platform: dmx512
    channel: 1
    universe: dmx
    id: dmx_1

  - platform: dmx512
    channel: 2
    universe: dmx
    id: dmx_2
    

  - platform: dmx512
    channel: 3
    universe: dmx
    id: dmx_3

  - platform: dmx512
    channel: 4
    universe: dmx
    id: dmx_4
    

  - platform: dmx512
    channel: 5
    universe: dmx
    id: dmx_5

  - platform: dmx512
    channel: 6
    universe: dmx
    id: dmx_6
    

  - platform: dmx512
    channel: 7
    universe: dmx
    id: dmx_7

  - platform: dmx512
    channel: 8
    universe: dmx
    id: dmx_8
    

  - platform: dmx512
    channel: 9
    universe: dmx
[... 1035 lines cut ...]
    red: dmx_109
    green: dmx_110
    blue: dmx_111
    white: dmx_112
    
 - platform: rgbw
   id: sizplatz1
   red: dmx_141
   green: dmx_142
   blue: dmx_143
   white: dmx_144    
    
  - platform: rgbw
    id: sizplatz2
    red: dmx_145
    green: dmx_146
    blue: dmx_147
    white: dmx_148
    
  - platform: rgbw
    id: sizplatz3
    red: dmx_149
    green: dmx_150
    blue: dmx_151
    white: dmx_152

andyboeh, 2022-01-31:
I updated your comment to include quotes around the code. Unfortunately, you didn't include the other relevant parts of your config like logger, external_components, any other components in use. Let's give it a try without it:

What do you mean with "About the log its a loop-detection."?

I think that nobody ever tried to address so many lights/channels. There are a few reasons I can think of, the most likely ones:

1. That's too many bytes to send in one loop iteration, thus it has to wait for the flush() and crashes. 
2. The logger is flooded with messages (debug output), can't cope with it and crashes.

Ad 1. I just pushed an update for this; You might have to add `refresh: 60s` to your `external_components` in order to use the new code right away:

```
external_components:
  - source: github://andyboeh/esphome-dmx512
    refresh: 60s
```

Ad 2. Please try to reduce the log level:
```
logger:
  level: ERROR
```

itsieber, 2022-02-01:
Hey Andy

Thx for the fast try.

Its not helps. The correct Error from Serial:

task watchdog got triggered. The followin tasks did not reset the watchdog in time:
tsk_wdt: - loopTask (CPU 1)

andyboeh, 2022-02-01:
What exactly did you try? 
Is your ESP doing something else?
Please **attach** your complete config (without the passwords, of course).

itsieber, 2022-02-02:
I had try the two options that your write. But when uncomment the few parts inget again the loop.

The Esp do only this functions on top. Nothing else.

andyboeh, 2022-02-02:
I don't have this many DMX devices, but I don't think it's necessary to try it out. If you could **attach** your complete config (without the passwords; not just the extract from before), I can replicate your setup and see what happens.

itsieber, 2022-02-13:
`
esphome:
  name: dmx-ug

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: ERROR
# Enable Home Assistant API
api:

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dmx-Ug Fallback Hotspot"
    password: ""

captive_portal:

external_components:
  - source: github://andyboeh/esphome-dmx512
    refresh: 60s

uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: 18
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: 18
  uart_num: 1

output:
  - platform: dmx512
    channel: 1
    universe: dmx
    id: dmx_1

  - platform: dmx512
    channel: 2
    universe: dmx
    id: dmx_2
    

  - platform: dmx512
    channel: 3
    universe: dmx
    id: dmx_3

  - platform: dmx512
    channel: 4
    universe: dmx
    id: dmx_4
    

  - platform: dmx512
    channel: 5
[... 1096 lines cut ...]
    blue: dmx_111
    white: dmx_112
    
  - platform: rgbw
    id: sizplatz1
    red: dmx_141
    green: dmx_142
    blue: dmx_143
    white: dmx_144    
    
  - platform: rgbw
    id: sizplatz2
    red: dmx_145
    green: dmx_146
    blue: dmx_147
    white: dmx_148
    
  - platform: rgbw
    id: sizplatz3
    red: dmx_149
    green: dmx_150
    blue: dmx_151
    white: dmx_152

`

andyboeh, 2022-02-13:
I edited your comment to attach the file - the comments are extremely long otherwise.
I'll investigate whenever I have the time to do so.

itsieber, 2022-02-14:
It's a ESP32-WROOM

itsieber, 2022-02-16:
its direct on the pinouts from dev wroom, i had also change the board, but nothing solved.

itsieber, 2022-03-16:
I had make many tests with other hardware.
Same Problem.

Interesting are there many reboots or lost connection.

After i add:
`power_save_mode: none`

Its mutch more stable.
So i have 2-4 times a fast reboot a day. all lights turn off.

On home assistent the lights are then "on" but in real there are off.
touch off/on and i have light again.

closed 2023-11-05 by andyboeh