andyboeh/esphome-dmx512 #12: 16bit depth fixtures

jsenecal, 2022-03-31:
Hi there,

I was wondering if there is a way to define 16bit outputs or to support that in the future. (for instance address 1 is coarse adjustment for RED, the second address is fine adjustment for RED)

andyboeh, 2022-03-31:
What already works is simply defining two channels, coarse_red and fine_red, for instance, and using both outputs to drive the fixture.
You can already achieve it by creating something like the following (completely untested):

```
uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: GPIO2 #ESP8266
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: GPIO2 #ESP8266
  uart_num: 1

output:
- platform: dmx512
  channel: 1
  universe: dmx
  id: coarse_red
- platform: dmx512
  channel: 2
  universe: dmx
  id: fine_red
- platform: template
  id: red_output
  type: float
  write_action:
    - lambda: |-
        int lvl = static_cast<int>(state);
        id(coarse_red).set_level((lvl & 0xff00) >> 8);
        id(fine_red).set_level(lvl & 0xff);
```

If you then use the output `red_output`, it should work as a 16 bit output.

jsenecal, 2022-03-31:
That is awesome, I'll test that and get back to you!

jsenecal, 2022-03-31:
I ended up with something similar, with few changes to account for the differences in scaling (0.0-1.0 vs 0-65535) :
```yaml
- platform: template
    id: d1_red_output
    type: float
    min_power: 0.01
    write_action:
      - lambda: |-
          int lvl = static_cast<int>(state*65534.0);
          uint8_t low = lvl & 0xff;
          uint8_t high = (lvl >> 8);
          id(d1_r_coarse).set_level(high/255.0);
          id(d1_r_fine).set_level(low/255.0);

```

closed 2022-03-31 by jsenecal

andyboeh, 2022-04-01:
Perfect, thanks for sharing your solution! I'll add it to the README / examples.

jsenecal, 2022-04-01:
Currently working on a way to "smooth out" the transitions between states, so that it doesn't just "jump" between two "coarse" states without incrementing the fine state gradually.