andyboeh/esphome-dmx512 #15: Unstable signal?

stijnb1234, 2022-09-26:
I'm using your component with my ESP32 Wrover kit. I've connected two lights to it, but one of the lights is flashing every second for a short periode. What can cause this?

I've connected the ESP32 to a MAX485 module as described in your README.

My ESPHome configuration:
```yml
external_components:
  - source: github://andyboeh/esphome-dmx512

esphome:
  name: esp-rgbdmx
  platform: esp32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "esp-p1reader"
    password: !secret fallback_password

captive_portal:

logger:
  baud_rate: 0

api:

ota:
  password: !secret ota_password

uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: GPIO1
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: GPIO1
  uart_num: 0

output:
  - platform: dmx512
    channel: 2
    universe: dmx
    id: lskon_red
  - platform: dmx512
    channel: 1
    universe: dmx
    id: lskon_green
  - platform: dmx512
    channel: 3
    universe: dmx
    id: lskon_blue
  - platform: dmx512
    channel: 4
    universe: dmx
    id: lskon_white
    
  - platform: dmx512
    channel: 5
    universe: dmx
    id: tdcsrig_master
  - platform: dmx512
    channel: 6
    universe: dmx
    id: tdcsrig_redyellow
  - platform: dmx512
    channel: 7
    universe: dmx
    id: tdcsrig_greenwhite
  - platform: dmx512
    channel: 8
    universe: dmx
    id: tdcsrig_bluepink
  - platform: dmx512
    channel: 9
    universe: dmx
    id: tdcsrig_strobe
  - platform: dmx512
    channel: 10
    universe: dmx
    id: tdcsrig_motor
  - platform: dmx512
    channel: 11
    universe: dmx
    id: tdcsrig_macro

light:
  - platform: rgbw
    name: LedstripKoning RGBW
    id: lskon_rgbw
    red: lskon_red
    green: lskon_green
    blue: lskon_blue
    white: lskon_white
    icon: mdi:spotlight
    default_transition_length: 0.3s
    color_interlock: false
    gamma_correct: 0
    restore_mode: RESTORE_DEFAULT_ON
  - platform: monochromatic
    name: Ayra TDC Sunrise Right
    output: tdcsrig_master
    icon: mdi:blur
    disabled_by_default: true
    default_transition_length: 2s
    gamma_correct: 0
    restore_mode: RESTORE_DEFAULT_ON
  - platform: rgb
    name: Ayra TDC Sunrise Right RGB
    id: tdcsrig
    red: tdcsrig_redyellow
    green: tdcsrig_greenwhite
    blue: tdcsrig_bluepink
    icon: mdi:spotlight
    default_transition_length: 0.3s
    gamma_correct: 0
    restore_mode: RESTORE_DEFAULT_ON
```

stijnb1234, 2022-09-26:
Verbose log when changing color:

```
[20:09:27][VV][api.service:458]: on_light_command_request: LightCommandRequest {
  key: 3228041757
  has_state: YES
  state: YES
  has_brightness: NO
  brightness: 0
  has_color_mode: YES
  color_mode: COLOR_MODE_RGB
  has_color_brightness: YES
  color_brightness: 1
  has_rgb: YES
  red: 1
  green: 0
  blue: 0
  has_white: NO
  white: 0
  has_color_temperature: NO
  color_temperature: 0
  has_cold_white: NO
  cold_white: 0
  has_warm_white: NO
  warm_white: 0
  has_transition_length: NO
  transition_length: 0
[20:09:27][D][light:035]: 'Ayra TDC Sunrise Right RGB' Setting:
[20:09:27][D][light:054]:   Color brightness: 100%
[20:09:27][D][light:058]:   Red: 100%, Green: 0%, Blue: 0%
[20:09:27][D][light:084]:   Transition length: 0.3s
[20:09:27][VV][api.service:122]: send_light_state_response: LightStateResponse {
  key: 3228041757
  state: YES
  brightness: 1
  color_mode: COLOR_MODE_RGB
  color_brightness: 1
  red: 1
  green: 0
  blue: 0
  white: 1
  color_temperature: 0
  cold_white: 1
  warm_white: 1
  effect: ''
}
[20:09:27][VV][esp32.preferences:039]: s_pending_save: key: 3228041757, len: 44
[20:09:27][D][dmx512:053]: write_channel 6: 0
[20:09:27][D][dmx512:053]: write_channel 7: 0
[20:09:27][D][dmx512:053]: write_channel 8: 255
[20:09:27][VV][uart.arduino_esp32:150]:     Flushing...
```

andyboeh, 2022-09-26:
I had similar problems with my dimmers and I invested like two full days in debugging it. It turned out that the light had a firmware bug that caused the flashing.

Why did you disable logging / UART0, i.e. why are you not using UART1 for DMX?
What happens if you set `update_interval` to `5000`, i.e. only update every 5 seconds?
What happens if you set `force_full_frames` to `true`?
What happens if you set `periodic_update` to `false`?

Please only try *one* of the options at once and report back what happens.

Apart from that, did you try the usual: Does it also flash with just one light connected? Does changing channels make any difference? Have you tried to disable WiFi?

stijnb1234, 2022-09-27:
Thanks for your response.

I've chosen UART0 because of the pins that are exposed, but I just found out UART1 is available on every GPIO pin.

I will try changing the values soon.

- I've tried connecting only one light, but the light with problems then still flashes.
- Changing cables did not fix anything. Tried long and short ones.
- Disabing wifi does not fix it.

stijnb1234, 2022-09-29:
- Changing to UART1 did nothing, as expected.
- Changing the `update_interval` to `5000` makes it turn on only for a short time every 5 seconds.
- Changing the `force_full_frames` to `true` makes it flicker worse (the off part is longer).
- Changing the `periodic_update` to `false` makes it turn off after changing the color.

andyboeh, 2022-09-30:
Interesting, so it turns off while receiving data - do you have another DMX controller you can try the problematic light with?
Did you try this light as the only light, configured to channel 0?
Which device (make and model) is it exactly?

stijnb1234, 2022-09-30:
- It turns off **after** receiving data. That's also causing the flickering, because it constantly updates. 
- I had another Dmx controller, and the light worked fine with that one. Sadly that controller is half broken now. 
- Yes, I tried it as the only light, but that didnt change it.
- It's the Ayra TDC Sunrise. Link to manual (in Dutch): https://www.bax-shop.nl/downloads/products/9000-0021-6269/ayra_tdc_sunrise_led_lighteffect_user_manual_edit.pdf

andyboeh, 2022-09-30:
Can you rule out cabling issues, i.e. length of the cable, cable termination?
Are you sure that the value for the master channel is always set to a valid value? You have it disabled by default with restore mode defaulting to on, does that really work as expected?

stijnb1234, 2022-09-30:
- No cables issues. Tried multiple Dmx (so not Xlr) cables, long and short ones. Sadly did nothing. Also have an Dmx terminator installed at the end.
- Yes. I tried enabled the master and setting it to 255 by hand, but the light still turns off (with periodic updates set to false) after changing the master state.

andyboeh, 2022-09-30:
One more thing to try: What happens if you set the `update_interval` to 0 (`periodic_update` needs to be `true` or unset)?

stijnb1234, 2022-09-30:
Wow, that fixed it! Thanks :)

closed 2022-09-30 by stijnb1234

andyboeh, 2022-09-30:
You're welcome! This may, however, limit the amount of work the ESP can do in the meantime since it spends quite a lot of time on the UART. You can play around with this value to find the threshold for your device.

What did we do: I designed the component in a way that new values are transmitted immediately while unchanged values are refreshed every 500ms. Obviously, your light decides to switch off if it doesn't receive a new DMX message within a shorter period. By setting the `periodic_update` to 0, we force the component to refresh the values as fast as possible.

I don't remember whether the standard mentions anything about the refresh, so I'm currently unsure who is violating the standard or if this is defined at all. My DMX dimmers, however, switch off after about 30 minutes (!) if they're not refreshed.

jcgruenhage, 2023-09-17:
I stumbled over this, as I also have a fixture which requires more frequent refreshes, and I wanted to know who's at fault here: The LED fixture is.

The spec specifies that a sender must send a refresh at least once per second, and that a receiver must be capable to handle a signal that refreshes at at least every 1.25 seconds. The default in this implementation of 2Hz/500ms is therefore sufficiently frequent, and fixtures shutting off after less than half a second without a signal aren't up to spec. But well, using 3 pin XLR is the common cabling for DMX and *also* prohibited by the spec, so apparently no one really cares about the spec anyway.

stijnb1234, 2023-09-17:
> I stumbled over this, as I also have a fixture which requires more frequent refreshes, and I wanted to know who's at fault here: The LED fixture is.
>
> The spec specifies that a sender must send a [...]

Yeah, I was expecting something like that. Anyway, happy that there was still a way to resolve this from the transmitter side.