andyboeh/esphome-dmx512 #17: i am not able to get it run

janosch1337, 2022-11-01:
Hey guys,

i have a azdelivery esp32 and a max3485 module... my dmx device is a 32channel 3A dmx512-decoder. so i ordered a usb to dmx cable to check if my led strip is working correctly - all ok (test it with qlc+).

then i connect the esp32 (pin5) to the max3485 and test it also...but nothing works..

i have set the dmx512 controller to adress 1 (at display) and turn it into 16bit mode. then connect a rgbw stripe to out1...in qlc+ its:

adress 1 : red
address 3: green
address 5 blue
address 7 white

    ```

external_components:
  - source: github://andyboeh/esphome-dmx512

uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: GPIO5
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: GPIO5
  enable_pin: GPIO33
  uart_num: 1
  periodic_update: true
  force_full_frames: false
  custom_break_len: 92
  custom_mab_len: 12
  update_interval: 500

light:
  - platform: monochromatic
    name: "Test Lights"
    output: coarse_red
    id: light_test
    default_transition_length: 2s

output:
- platform: dmx512
  channel: 1
  universe: dmx
  id: coarse_red
```

maybe you can help me ;)

andyboeh, 2022-11-01:
16 bit mode is not supported without a lambda - see the example on the README.

janosch1337, 2022-11-02:
Yes.. Already tested..

andyboeh, 2022-11-02:
It's a bit hard to help you if you do not provide such details. The config you posted above definitely doesn't work with a controller in 16 bit mode. Please post the config you have tried so that we can see whether it is your config.
Please provide a few more hardware details: Did you double-check the wiring? What exactly did you wire? Which modules did you use exactly?

janosch1337, 2022-11-02:
ok sorry.. try to give more details..

azdelivery esp32  connected with 3,3v and gnd to max3485 (https://www.amazon.de/DollaTek-max3485-entwicklung-konverter-entfernbarer/dp/B08HQMDKGR/ref=sr_1_5?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1AZKFZUM90A37&keywords=max3485&qid=1667378146&qu=eyJxc2MiOiIzLjM4IiwicXNhIjoiMi41MCIsInFzcCI6IjIuNDUifQ%3D%3D&sprefix=max3485%2Caps%2C124&sr=8-5)

gpio5 connect to RCV pin of max3485 (also tried gpio17 or 16... and tried tx pin of max3485)...

my dmx512 controller is able to switch between 8 bit and 16bit...so i tried this also...

`substitutions:
  device_name: dmx-test
  friendly_name: DMX Test
  device_description: DMX esp32
  
esphome:
  name: ${device_name}
  comment: ${device_description}
  platform: ESP32
  board: esp32dev

# WiFi connection
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
   ssid: ${device_name}_fallback
   password: !secret ap_password
   ap_timeout: 1min
  use_address: ${device_name}.local
  #manual_ip:
  #  static_ip: 192.168.1.20
  #  gateway: 192.168.1.1
  #  subnet: 255.255.255.0

logger:

# Enable Home Assistant API
api:
  password: !secret api_password

# Enable over-the-air updates
ota:
  password: !secret ota_password

# Enable Web server
web_server:
  port: 80

# Sync time with Home Assistant
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information
text_sensor:
  - platform: version
    name: ${friendly_name} Version

sensor:
  # Uptime sensor
  - platform: uptime
    name: ${friendly_name} Uptime
    
external_components:
  - source: github://andyboeh/esphome-dmx512

uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: GPIO5
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: GPIO5
  enable_pin: GPIO33
  uart_num: 2
  periodic_update: true
  force_full_frames: false
  custom_break_len: 92
  custom_mab_len: 12
  update_interval: 500

light:
  - platform: monochromatic
    name: "Test Lights"
    output: red_output
    id: light_test
    default_transition_length: 2s

output:
- platform: dmx512
  channel: 1
  universe: dmx
  id: coarse_red
- platform: dmx512
  channel: 2
  universe: dmx
  id: fine_red
- platform: template
  id: red_output
  type: float
  min_power: 0.01
  write_action:
    - lambda: |-
        int lvl = static_cast<int>(state*65535.0);
        uint8_t low = lvl & 0xff;
        uint8_t high = (lvl >> 8);
        id(coarse_red).set_level(high/255.0);
        id(fine_red).set_level(low/255.0);`

https://www.amazon.de/Decoder-Controller-Channels-Controllder-DC5V-24V/dp/B075HC8481/ref=sr_1_6?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=18XKO5G1MK11N&keywords=dmx512+controller&qid=1667378358&qu=eyJxc2MiOiI0LjM3IiwicXNhIjoiMy4yMyIsInFzcCI6IjIuODYifQ%3D%3D&sprefix=dmx512+controlle%2Caps%2C79&sr=8-6

janosch1337, 2022-11-02:
now the dmx512 decoder gets some data... but the leds are only blinking....it seems so that the esp sends data and then the data send stops...and again

janosch1337, 2022-11-02:
so ok now its working... my fault :D had usb from computer with qlc open plugged in the same time with esp32 :-D so qlc resets what esp32 send ;)

thank you for your support!!

```
dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: GPIO5
  #enable_pin: GPIO33
  uart_num: 1
  periodic_update: false
  force_full_frames: true
  custom_break_len: 92
  custom_mab_len: 12
  update_interval: 0

light:
  - platform: monochromatic
    name: "Test Lights"
    output: coarse_red
    id: light_test
    default_transition_length: 2s

output:
- platform: dmx512
  channel: 1
  universe: dmx
  id: coarse_red
```

this is now working for 8bit. and red on first channel. somtimes the led stripe is blinking in the first 2s of transition

andyboeh, 2022-11-02:
One more thing that should improve your situation: `force_full_frames: false` and `periodic_update: true`. This way you might be able to increase the `update_interval`. `force_full_frames` is very expensive when it comes to processing time of the ESP32, so enable it only if it's really required.

closed 2022-11-02 by andyboeh