andyboeh/esphome-dmx512 #25: Possible to output DMX directly?

bucovaina, 2023-10-03:
Hi

**My use case:**
I'm using your custom component to my content on an ESP32 WT32-ETH01 device which drives a 12 channel dimmer capable of delivering 43kW to 12 channels. I'm "only" using 4 channels with a combined power output of 12.4kW. 3 channels are for a 3 phase flow heater, 10kW and 1 channel for an electric boiler. Now Node-Red and Home Assistant are used to give DMX512 signals to the ESP32. But I want more reliability and robustness. So I'd like to take Node-Red and Home Assistant out of the mix if possible at all. Doing so I'd avoid possibly wrong dim levels to be set if eg. an upgrade fails.

**My question:**
Is it possible with automations and lambda functions to let esphome itself write directly to DMX512? Avoiding Node-Red/HOme Assistant as a dependency. If this is possible, could you give me an example how to do it? I'm a bit puzzled since I'm a bit new to esphome.

Thanks a lot.

andyboeh, 2023-10-03:
Sure: Check out the docs, you can use the `light.turn_on` call as well as the `light.dim_relative` call, whatever suits your needs: https://esphome.io/components/light/index.html#light-turn-on-action

bucovaina, 2023-10-03:
Ha I read the docs but didn't realize it was going to be this straightforward, feel free to close this issue, I guess I'll be able to figure it out :)

closed 2023-11-05 by andyboeh