andyboeh/esphome-dmx512 #26: Possible entities for HA

hibbeldi, 2023-11-05:
Sorry I'm a noob.
I am trying to unterstand the project (don't want to order all stuff without an idea how to integrate).

A) Can I include a DMX touch controller with this (has both lan and 3 pin (could use both in general) (DMX input)

B) A dimpack (3 pin)  (seems straightforward)

I already orded the 485Max ànd have esps running on ha (mainly for relays).

1000 thanks for your work

andyboeh, 2023-11-05:
Have you read the first sentence of the project description? 

> This is a DMX512 custom component for ESPHome that allows Arduino-based devices to control DMX devices via UART (this requires an RS485 module, see below).

So obviously:
ad A) No. It's an output component, not an input component. It turns your ESP into a DMX controller.
ad b) There is no question involved. Can you control it? Yes, that's the purpose.

hibbeldi, 2023-11-05:
Many thanks for your flash reply!!
(Sorry right self explanatory, had some hope that this is softwarewise maybe the same ;o)
If you now of any exisiting library for that it would very kind if you could give me a hint.

andyboeh, 2023-11-05:
No, software-wise this is quite different as reading DMX is not exactly easy due to the timing constraints of the protocol (not impossible, though). 

According to [this](https://community.home-assistant.io/t/dmx-lighting/2248/348), the repository at https://github.com/Breina/ha-artnet-led allows DMX input using an ArtNet Node. You need an ArtNet node for this.

closed 2023-11-12 by hibbeldi