andyboeh/esphome-dmx512 #27: Build on ESP8266 fails: SET_PERI_REG_MASK' was not declared

psxde, 2023-12-03:
Trying to compile with esp8266 end in:

```
Compiling .pioenvs\led-buero-dmx\src\esphome\components\light\light_state.cpp.o
src\esphome\components\dmx512\dmx512esp8266.cpp: In member function 'virtual void esphome::dmx512::DMX512ESP8266::send_break()':
src\esphome\components\dmx512\dmx512esp8266.cpp:13:5: error: 'SET_PERI_REG_MASK' was not declared in this scope
   13 |     SET_PERI_REG_MASK(UART_CONF0(this->uart_idx_), UART_TXD_BRK);
      |     ^~~~~~~~~~~~~~~~~
src\esphome\components\dmx512\dmx512esp8266.cpp:15:5: error: 'CLEAR_PERI_REG_MASK' was not declared in this scope
   15 |     CLEAR_PERI_REG_MASK(UART_CONF0(this->uart_idx_), UART_TXD_BRK);
      |     ^~~~~~~~~~~~~~~~~~~
*** [.pioenvs\led-buero-dmx\src\esphome\components\dmx512\dmx512esp8266.cpp.o] Error 1
src\esphome\components\light\light_call.cpp:11:25: warning: 'const esphome::LogString* esphome::light::color_mode_to_human(esphome::light::ColorMode)' defined but not used [-Wunused-function]
   11 | static const LogString *color_mode_to_human(ColorMode color_mode) {
      |                         ^~~~~~~~~~~~~~~~~~~
============================================== [FAILED] Took 7.55 seconds ==============================================

```

Do you have any ideas?

Thanks,
Patrick

andyboeh, 2023-12-04:
I never test on ESP8266, so it's likely that some change in ESPHome causes the build failure.  
What is your ESPHome version?

psxde, 2023-12-04:
I have tried multiple versions, including 2023.8.3 and 2023.11.6, and it fails every time. Works fine with ESP32.

The yaml:
```
esphome:
  name: led-buero-dmx

esp8266:
  board: d1_mini

# WiFi
wifi:
  ssid: !secret home_wifi_ssid
  password: !secret home_wifi_password

# DMX Stuff
external_components:
  - source: github://andyboeh/esphome-dmx512

uart:
  id: uart_bus
  baud_rate: 250000
  tx_pin: GPIO2 # D4
  stop_bits: 2

dmx512:
  id: dmx
  uart_id: uart_bus
  tx_pin: GPIO2 # D4
  uart_num: 1

# LED RGB+CCT-Panel
output:
  - platform: dmx512
    channel: 1
    universe: dmx
    id: dmx_r
  - platform: dmx512
    channel: 2
    universe: dmx
    id: dmx_g
  - platform: dmx512
    channel: 3
    universe: dmx
    id: dmx_b
  - platform: dmx512
    channel: 4
    universe: dmx
    id: dmx_cw
  - platform: dmx512
    channel: 5
    universe: dmx
    id: dmx_ww

light:
  - platform: rgbww
    id: panel
    name: panel
    red: dmx_r
    green: dmx_g
    blue: dmx_b
    cold_white: dmx_cw
    warm_white: dmx_ww
    cold_white_color_temperature: 153 mireds
    warm_white_color_temperature: 500 mireds
    constant_brightness: true
    color_interlock: true
    gamma_correct: 0
```

andyboeh, 2023-12-04:
Fixed in https://github.com/andyboeh/esphome-dmx512/commit/dbd89b491bc1de43818cf03ac80418cdbbe47ba6. If you want to test it today, you will have to change your YAML to
```
external_components:
  - source: github://andyboeh/esphome-dmx512
    refresh: 60s
```

This forces a recheck of the remote components if the last update was more than 60s ago.

psxde, 2023-12-04:
Thank you very much, the build is now successful :)

closed 2023-12-04 by psxde