andyboeh/esphome-dmx512 #31: Lambda for dimmer

stijnb1234, 2024-02-27:
I have a DMX dimmer in my circuit with weird values:
Below 124 is off
124 - 150 is dimming from 0-100%
Above 150 is on

How can I do this with this component? Can I use the lambda, and how?

andyboeh, 2024-02-28:
Sure: Create a template output and add your logic to the `write_value` function:

https://esphome.io/components/output/template.html

stijnb1234, 2024-02-28:
So far I understand. But what are the input values (which range?) and what are the output values?

andyboeh, 2024-02-28:
The values at the output are in the range 0.0 to 1.0, it's a float output. The DMX component calculates the 0-254 values based on this float range. So you are dealing with two float ranges between 0 and 1 here and you need to map your values here.

closed 2024-12-02 by andyboeh