andyboeh/esphome-dmx512 #34: passthrough a 'raw' 0-255 value

wauswaus, 2025-03-05:
Hi,

First of all, thanks for your work!
I'm using this component to control a heating element through a DMX dimmer.
The issue i'm facing is that when I run it via the monochromatic platform, the dimming values are not linear (due to gamma correction I believe by the monochromatic component).
For instance, if is set the dim value to 10%, a value of 0 is sent to the dimmer:
[20:57:07][D][light:051]:   Brightness: 10%
[20:57:07][D][dmx512:053]: write_channel 1: 0

[20:57:59][D][light:051]:   Brightness: 20%
[20:57:59][D][dmx512:053]: write_channel 1: 2

Is there a way I can sent a value from 0-255 directly to the dmx dimmer?

andyboeh, 2025-03-05:
You should be able to configure it as an output and use the output directly. Something like this:

```
output:
- platform: dmx512
  channel: 1
  universe: dmx
  id: dimmer
```

Then you can set the value with output.set_level. This is still in the 0.0 - 1.0 range, but at least no gamma correction should be applied.

wauswaus, 2025-03-05:
Thanks for your quick answer. I probably need a template output that drives dmx512 output. I'll give it a crack this weekend.
I did search to see if I could disable the gamma correction but couldn't find anything. How should I do this? (or is at as simple as setting the correction to 0).

closed 2025-03-05 by wauswaus