andyboeh/esphome-dmx512 #40: logger

defl, 2025-09-16:
I'm using a [Kincony A6](https://www.kincony.com/esp32-6-channel-relay-module-kc868-a6.html) which has a ESP32-WROOM-32 and an RS-485 interface to drive a DMX bus via ESPHome. For the longest time I had issues with DMX "not working" so I decided to dig in a bit more.

I started out the the following relevant bits in my config:

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf

logger:
  baud_rate: 0  # 0 = disable
    
uart:

  - id: rs485_bus
    tx_pin: 
      number: GPIO27
      allow_other_uses: true
    rx_pin: GPIO14
    baud_rate: 250000
    stop_bits: 2

dmx512:
  id: dmx
  uart_id: rs485_bus
  tx_pin: 
    number: GPIO27
    allow_other_uses: true
  uart_num: 1
  update_interval: 100
  periodic_update: true
  force_full_frames: false
  custom_break_len: 140 
  custom_mab_len: 14
```

This gets me the following waveform:
[image]

Notice that we're missing the  MB and MAB here? I've tried the following things:
- Different framework (switch back to arduino)
- Various different custom break and mab lengths
- Switch uarts in the dmx512 config

Some of these combination get me only what looks like the MB but I'm not quite sure what it is. After some experimentation and stripping everything out that I could it turned out that the logger logger here was the culprit. I generally run with serial logging as I don't use it by setting baud_rate: 0. 

If I remove that statement, which enables serial logging again (UART0), that somehow enables the UART I configure in the DMX component (either 1 or 2, both work) and all of a sudden I have MB and MAB back:
[image]

Maybe something to check or note in the docs for the next poor soul that spends an evening trying to debug esphome DMX.

andyboeh, 2025-10-01:
I'm quite sure that you need to set `uart_num` to 0 in this configuration as UART0 is not used by the logger. The UART number is required for the generation of the break signal, in your case, it was probably "sent" via UART1.
Re-enabling the logger made the DMX code use UART1 and hence it started working.

Could you test this configuration:

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf

logger:
  baud_rate: 0  # 0 = disable
    
uart:

  - id: rs485_bus
    tx_pin: 
      number: GPIO27
      allow_other_uses: true
    baud_rate: 250000
    stop_bits: 2

dmx512:
  id: dmx
  uart_id: rs485_bus
  tx_pin: 
    number: GPIO27
    allow_other_uses: true
  uart_num: 0
```

defl, 2025-10-06:
Thanks for the reply @andyboeh, I can confirm that works too. I know how you can do to the break with UART and then switch over to the UART, did something similar on a STM32 a while ago, Didn't dive into the code but why does the uart_num _have_ to be zero here to work? I assigned it uart_num 1 with the assumption that that would work.

andyboeh, 2025-10-06:
Because that's not an assignment. You tell the DMX component which internal UART number the `uart` component has been assigned so that the DMX component can generate the break signal. That's a hack and only required because there is no way to retrieve the internal UART number from within the component itself. At the same time, this is the main reason why I am not going to submit this component upstream.

The UART ID is auto-assigned by ESPHome, if you use the logger, the logger gets UART0 and the first additional UART will be UART1. However, if the logger is disabled, the UART component will be UART0.

closed 2025-10-07 by defl